mavlink/mavlink2rest #7: Thread panic when changing flight mode

NickNothom, 2020-05-25:
Running the stock Ardusub Pi Image, with the Mavlink2Rest 0.6.0 release binary. Vehicle is using ArduRover 4.0 Stable. 

When I change the flight mode (in QGC), the Mavlink2Rest server crashes with the following output

```
pi@champ:~/companion/tools $ ./mavlink2rest --connect udpin:127.0.0.1:9002 --server 0.0.0.0:4777
MAVLink connection string: udpin:127.0.0.1:9002
REST API address: 0.0.0.0:4777
thread '<unnamed>' panicked at 'Unexpected enum value 11.', /target/armv7-unknown-linux-musleabihf/release/build/mavlink-c1b7c2ed7a6a321a/out/common.rs:4124:13
stack backtrace:
   0:   0x60a6c4 - backtrace::backtrace::libunwind::trace::h2740b7779bb3fbf6
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunwind.rs:88
   1:   0x60a6c4 - backtrace::backtrace::trace_unsynchronized::hc587e09a69bdc583
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs:66
   2:   0x60a6c4 - std::sys_common::backtrace::_print_fmt::hde9114c81a02c611
                       at src/libstd/sys_common/backtrace.rs:77
   3:   0x60a6c4 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h069b16b4a6ea2d7a
                       at src/libstd/sys_common/backtrace.rs:59
   4:   0x6307b0 - core::fmt::write::h3fe08fe0e1c8dec3
                       at src/libcore/fmt/mod.rs:1052
   5:   0x60508c - std::io::Write::write_fmt::hb40c902bf8e5da41
                       at src/libstd/io/mod.rs:1426
   6:   0x60cb3c - std::sys_common::backtrace::_print::hc4e20dd38a1f8083
                       at src/libstd/sys_common/backtrace.rs:62
   7:   0x60cb3c - std::sys_common::backtrace::print::hf29b05d8b728a743
                       at src/libstd/sys_common/backtrace.rs:49
   8:   0x60cb3c - std::panicking::default_hook::{{closure}}::h2c253e0d3587e013
                       at src/libstd/panicking.rs:204
   9:   0x60c7b0 - std::panicking::default_hook::h637f76a8fdb3ad07
                       at src/libstd/panicking.rs:224
  10:   0x60d1ec - std::panicking::rust_panic_with_hook::h2dccb74573dc29e8
                       at src/libstd/panicking.rs:472
  11:   0x60ce00 - rust_begin_unwind
                       at src/libstd/panicking.rs:380
  12:   0x62f064 - core::panicking::panic_fmt::hc9fedf85d2814a69
                       at src/libcore/panicking.rs:85
  13:   0x62ee44 - core::option::expect_failed::hb19b537baead1f0a
                       at src/libcore/option.rs:1191
  14:   0x40d764 - mavlink::common::COMMAND_ACK_DATA::deser::h72a1b199dc63d25b
  15:   0x4284ec - <mavlink::common::MavMessage as mavlink::Message>::parse::hfbe47a60d6f17d16
  16:   0x320a8c - mavlink::read_versioned_msg::h58b4e622963d9ac3
  17:   0x30749c - <mavlink::connection::udp::UdpConnection as mavlink::connection::MavConnection<M>>::recv::h351d371a76729332
  18:   0x3f6e10 - std::sys_common::backtrace::__rust_begin_short_backtrace::h851d5b082366ffcc
  19:   0x31d090 - std::panicking::try::do_call::hb53df87ce34ece45
  20:   0x610638 - __rust_maybe_catch_panic
                       at src/libpanic_unwind/lib.rs:86
  21:   0x32908c - core::ops::function::FnOnce::call_once{{vtable.shim}}::h71edd6a2fe43f6d5
  22:   0x5ff568 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::haa16aef56df3b543
                       at /rustc/b8cedc00407a4c56a3bda1ed605c6fc166655447/src/liballoc/boxed.rs:1015
  23:   0x60fec8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::hddbf5f21bb2e7945
                       at /rustc/b8cedc00407a4c56a3bda1ed605c6fc166655447/src/liballoc/boxed.rs:1015
  24:   0x60fec8 - std::sys_common::thread::start_thread::ha699d94be8e2f96d
                       at src/libstd/sys_common/thread.rs:13
  25:   0x60fec8 - std::sys::unix::thread::Thread::new::thread_start::hd01e2c359672fbae
                       at src/libstd/sys/unix/thread.rs:80

```

And with RUST_BACKTRACE=full:

```
pi@champ:~/companion/tools $ ./mavlink2rest --connect udpin:127.0.0.1:9002 --server 0.0.0.0:4777
MAVLink connection string: udpin:127.0.0.1:9002
REST API address: 0.0.0.0:4777
thread '<unnamed>' panicked at 'Unexpected enum value 11.', /target/armv7-unknown-linux-musleabihf/release/build/mavlink-c1b7c2ed7a6a321a/out/common.rs:4124:13
stack backtrace:
   0:   0x60a6c4 - backtrace::backtrace::libunwind::trace::h2740b7779bb3fbf6
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunwind.rs:88
   1:   0x60a6c4 - backtrace::backtrace::trace_unsynchronized::hc587e09a69bdc583
                       at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs:66
   2:   0x60a6c4 - std::sys_common::backtrace::_print_fmt::hde9114c81a02c611
                       at src/libstd/sys_common/backtrace.rs:77
   3:   0x60a6c4 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h069b16b4a6ea2d7a
                       at src/libstd/sys_common/backtrace.rs:59
   4:   0x6307b0 - core::fmt::write::h3fe08fe0e1c8dec3
                       at src/libcore/fmt/mod.rs:1052
   5:   0x60508c - std::io::Write::write_fmt::hb40c902bf8e5da41
                       at src/libstd/io/mod.rs:1426
   6:   0x60cb3c - std::sys_common::backtrace::_print::hc4e20dd38a1f8083
                       at src/libstd/sys_common/backtrace.rs:62
   7:   0x60cb3c - std::sys_common::backtrace::print::hf29b05d8b728a743
                       at src/libstd/sys_common/backtrace.rs:49
   8:   0x60cb3c - std::panicking::default_hook::{{closure}}::h2c253e0d3587e013
                       at src/libstd/panicking.rs:204
   9:   0x60c7b0 - std::panicking::default_hook::h637f76a8fdb3ad07
                       at src/libstd/panicking.rs:224
  10:   0x60d1ec - std::panicking::rust_panic_with_hook::h2dccb74573dc29e8
                       at src/libstd/panicking.rs:472
  11:   0x60ce00 - rust_begin_unwind
                       at src/libstd/panicking.rs:380
  12:   0x62f064 - core::panicking::panic_fmt::hc9fedf85d2814a69
                       at src/libcore/panicking.rs:85
  13:   0x62ee44 - core::option::expect_failed::hb19b537baead1f0a
                       at src/libcore/option.rs:1191
  14:   0x40d764 - mavlink::common::COMMAND_ACK_DATA::deser::h72a1b199dc63d25b
  15:   0x4284ec - <mavlink::common::MavMessage as mavlink::Message>::parse::hfbe47a60d6f17d16
  16:   0x320a8c - mavlink::read_versioned_msg::h58b4e622963d9ac3
  17:   0x30749c - <mavlink::connection::udp::UdpConnection as mavlink::connection::MavConnection<M>>::recv::h351d371a76729332
  18:   0x3f6e10 - std::sys_common::backtrace::__rust_begin_short_backtrace::h851d5b082366ffcc
  19:   0x31d090 - std::panicking::try::do_call::hb53df87ce34ece45
  20:   0x610638 - __rust_maybe_catch_panic
                       at src/libpanic_unwind/lib.rs:86
  21:   0x32908c - core::ops::function::FnOnce::call_once{{vtable.shim}}::h71edd6a2fe43f6d5
  22:   0x5ff568 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::haa16aef56df3b543
                       at /rustc/b8cedc00407a4c56a3bda1ed605c6fc166655447/src/liballoc/boxed.rs:1015
  23:   0x60fec8 - <alloc::boxed::Box<F> as core::ops::function::FnOnce<A>>::call_once::hddbf5f21bb2e7945
                       at /rustc/b8cedc00407a4c56a3bda1ed605c6fc166655447/src/liballoc/boxed.rs:1015
  24:   0x60fec8 - std::sys_common::thread::start_thread::ha699d94be8e2f96d
                       at src/libstd/sys_common/thread.rs:13
  25:   0x60fec8 - std::sys::unix::thread::Thread::new::thread_start::hd01e2c359672fbae
                       at src/libstd/sys/unix/thread.rs:80

```

patrickelectric, 2020-05-25:
Thanks @NickNothom I'll try to replicate it here.

patrickelectric, 2020-05-25:
This problems is from both ArduPilot and QGC:
- It appears that QGC is using the deprecated message [SET_MODE](https://mavlink.io/en/messages/common.html#SET_MODE). Reference: [code](https://github.com/mavlink/qgroundcontrol/blob/ac08dd5a25700d29bd690a2d4bc6d8e915f8bc01/src/Vehicle/Vehicle.cc#L2523)
- And ardupilot is sending a wrong value for command in [COMMAND_ACK](https://mavlink.io/en/messages/common.html#COMMAND_ACK). Reference: [code](https://github.com/ardupilot/ardupilot/blob/5100c9fb8c3b2688a6717e4330fbeaf3fcf3e85a/libraries/GCS_MAVLink/GCS_Common.cpp#L2083)

This results in a terrible problem, where both applications are broken the mavlink spec.

patrickelectric, 2020-05-25:
Hi @NickNothom, Can you check the last release ?
https://github.com/patrickelectric/mavlink2rest/releases/tag/0.6.1

NickNothom, 2020-06-01:
Appears to be working! Thanks

closed 2020-06-01 by patrickelectric