mavlink/mavlink2rest #17: Not working on RPi Zero

Dronotique, 2020-07-02:
Hello,

I try to install on a RPi Zero (ARMv6).

If I download mavlink2rest-armv7-unknown-linux-musleabihf file (ARMv7) I got an error : `Segmentation fault`

If I try to install it with cargo I got this error (same error using sudo) : 
`error: failed to compile `mavlink2rest v0.7.3`, intermediate artifacts can be found at `/tmp/cargo-installK7vrEl`

Caused by:
  failed to parse lock file at: /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/mavlink2rest-0.7.3/Cargo.lock

Caused by:
  invalid serialized PackageId for key `package.dependencies`
`
I tried to remove this file and execute again, but I got a lot of compilation error 

`error[E0658]: use of unstable library feature 'alloc': this library is unlikely to be stabilized in its current form or name (see issue #27783)
  --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/lib.rs:41:1
   |
41 | extern crate alloc;
   | ^^^^^^^^^^^^^^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
 --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:5:23
  |
5 | use core::mem::{self, MaybeUninit};
  |                       ^^^^^^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2312:20
     |
2312 |     rules_storage: MaybeUninit<[(Register, RegisterRule<R>); MAX_RULES]>,
     |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2329:28
     |
2329 |             rules_storage: MaybeUninit::uninit(),
     |                            ^^^^^^^^^^^^^^^^^^^

error[E0599]: no function or associated item named `uninit` found for type `core::mem::MaybeUninit<_>` in the current scope
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2329:41
     |
2329 |             rules_storage: MaybeUninit::uninit(),
     |                            -------------^^^^^^
     |                            |
     |                            function or associated item not found in `core::mem::MaybeUninit<_>`

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2366:60
     |
2366 |             core::slice::from_raw_parts(self.rules_storage.as_ptr() as *const _, self.rules_len)
     |                                                            ^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2374:36
     |
2374 |                 self.rules_storage.as_mut_ptr() as *mut _,
     |                                    ^^^^^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2405:50
     |
2405 |                     let ptr = self.rules_storage.as_mut_ptr() as *mut (Register, RegisterRule<R>);
     |                                                  ^^^^^^^^^^

error[E0658]: use of unstable library feature 'maybe_uninit' (see issue #53491)
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/gimli-0.21.0/src/read/cfi.rs:2446:42
     |
2446 |             let ptr = self.rules_storage.as_mut_ptr() as *mut (Register, RegisterRule<R>);
     |                                          ^^^^^^^^^^

error: aborting due to 9 previous errors

Some errors occurred: E0599, E0658.
For more information about an error, try `rustc --explain E0599`.
error: failed to compile `mavlink2rest v0.7.3`, intermediate artifacts can be found at `/tmp/cargo-installeZYCIG`

Caused by:
  Could not compile `gimli`.`

patrickelectric, 2020-07-02:
Hi @Dronotique, thanks for opening this issue.
I can test it during the weekend, but I have created a armv6 build in my continuous CI system, you can check the binaries available here: https://github.com/patrickelectric/mavlink2rest/releases/tag/continuous
The [mavlink2rest-arm-unknown-linux-musleabihf](https://github.com/patrickelectric/mavlink2rest/releases/download/continuous/mavlink2rest-arm-unknown-linux-musleabihf) is armv6, let me know if it works for you.

patrickelectric, 2020-07-02:
Also, this errors messages are a bit weird, how did you install cargo ?
I recommend to use this: https://rustup.rs/
It appears that you are using an old version, rust updates quickly and debian based distros are really slow to keep track, the recommend way to install Rust is via rustup.

Dronotique, 2020-07-04:
Hello,

I'm in vacancy for one week. I'll test the build when I came back work.

I installés cargo with Apt, then it should bé the probem (I'm famillar with cargo ans rust)

Thank you very much!

patrickelectric, 2020-07-04:
@Dronotique Ok, let me know how it goes, I'll be out for vacation from july 10 until july 21, we are probably going to be out of sync. 
Hopefully this armv6 release will fix your problem.

Dronotique, 2020-07-04:
I'll try when I come back.
I'll get you in touch

patrickelectric, 2020-07-29:
Hi @Dronotique any update ?

Dronotique, 2020-07-29:
Sorry, I had a lot of work and now I'm mooving UP.
This task IS in m'y next todo list.
I think I will have a try next week

patrickelectric, 2020-07-29:
great, thanks for the update!

closed 2020-08-20 by Dronotique

patrickelectric, 2020-08-20:
No problem, let me know if any feature is necessary or any bug need to be fixed, be free to open new issues.