mavlink/mavlink2rest #19: unsupported message

arendeutsch, 2020-08-25:
Hello, when trying to send a set mode i get unsupported command.
I've tried using both that message body
{
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "SET_MODE",
            "custom_mode": 2,
            "target_system": 0,
            "base_mode": {
                "type": "MAV_MODE_PREFLIGHT"
            }
        }
    }

and this one
{
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 2,  # mode
            "param2": 0.0,  # custom mode
            "param3": 0.0,  # custom submode
            "param4": 0.0,  # empty
            "param5": 0.0,  # empty
            "param6": 0.0,  # empty
            "param7": 0.0,  # empty
            "command": {
                "type": "MAV_CMD_DO_SET_MODE"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}

Any suggestion ?

patrickelectric, 2020-08-25:
Hi @arendeutsch, Please can you inform which version of mavlink2rest that you are running ? It should give you a feedback about wrong messages.
Both commands work just fine here, can you please information what is the error message ?

patrickelectric, 2020-08-25:
If you want to try it yourself:
```bash
$ export DATA='{                                                                                                                                                                                          
        "header": {                                                                                                                                                                                                 
            "system_id": 1,                                                                                                                                                                                         
            "component_id": 1,                                                                                                                                                                                      
            "sequence": 0                                                                                                                                                                                           
        },                                                                                                                                                                                                          
        "message": {                                                                                                                                                                                                
            "type": "SET_MODE",                                                                                                                                                                                     
            "custom_mode": 2,                                                                                                                                                                                       
            "target_system": 0,                                                                                                                                                                                     
            "base_mode": {                                                                                                                                                                                          
                "type": "MAV_MODE_PREFLIGHT"                                                                                                                                                                        
            }                                                                                                                                                                                                       
        }                                                                                                                                                                                                           
    }'                                                                                                                                                                                                              
$ curl --request POST http://0.0.0.0:8088/mavlink -H "Content-Type: application/json" --data $DATA
```

or

```bash
$ export DATA='{
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 2,
            "param2": 0.0,
            "param3": 0.0,
            "param4": 0.0,
            "param5": 0.0,
            "param6": 0.0,
            "param7": 0.0,
            "command": {
                "type": "MAV_CMD_DO_SET_MODE"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}'
$ curl --request POST http://0.0.0.0:8088/mavlink -H "Content-Type: application/json" --data $DATA
```

arendeutsch, 2020-08-25:
I'm using version 0.8.0 of mavlink2rest

patrickelectric, 2020-08-25:
Please provide the output of the commands that I just said in my previous post or the answer from the server when you do a http post.

arendeutsch, 2020-08-25:
curl --request POST http://0.0.0.0:4777/mavlink -H "Content-Type: application/json" --data $DATA
Error: ["Failed to parse common message: Error(\"EOF while parsing a value\", line: 1, column: 10)", "Failed to parse ardupilotmega message: Error(\"EOF while parsing a value\", line: 1, column: 10)"]curl: (3) [globbing] unmatched brace in column 1
curl: (6) Could not resolve host: 1,
curl: (6) Could not resolve host: "component_id"
curl: (6) Could not resolve host: 1,
curl: (6) Could not resolve host: "sequence"
curl: (3) [globbing] unmatched close brace/bracket in column 2
curl: (6) Could not resolve host: "message"
curl: (3) [globbing] unmatched brace in column 1
curl: (6) Could not resolve host: "SET_MODE",
curl: (6) Could not resolve host: "custom_mode"
curl: (6) Could not resolve host: 2,
curl: (6) Could not resolve host: "target_system"
curl: (6) Could not resolve host: 0,
curl: (6) Could not resolve host: "base_mode"
curl: (3) [globbing] unmatched brace in column 1
curl: (3) [globbing] unmatched close brace/bracket in column 21

patrickelectric, 2020-08-25:
It appears that something is wrong with your bash or curl, it's failing to create the command and contact with the server, as you can see from the error message, it's ignoring the http:// address and trying to use everything after that as a host address.

Please provide the output of `curl --request GET http://0.0.0.0:4777/mavlink ` and `curl --version`.

Besides that, can you provide a screenshot of both `export DATA=...` and `curl --request POST..` to see if anything was copy-pasted wrongly ?

arendeutsch, 2020-08-25:
[image]

patrickelectric, 2020-08-25:
Hi @arendeutsch, thank you for the information, it appears that this version of curl has a problem related to the data field, I'm running 7.71 here.
To fix that problem in your curl version, please call `curl` `$DATA` variable with double quotes, line this:
`curl --request POST http://0.0.0.0:4777/mavlink -H "Content-Type: application/json" --data "$DATA" `

[image]

arendeutsch, 2020-08-26:
Thank. I get now empty response which i guess means OK 200.
But nothing happens. I was expected to see the flight mode changes to Depth hold mode.

BTW - The same thing is was happening yesterday when i used postman and when i was running it via python script using post requests. I get response OK 200. but when looking at the log of mavproxy i get "God mavlink message COMMAND_ACK {command: 176, result: 3}" which means unsupported

arendeutsch, 2020-08-26:
Just an update on my testing.
When using the following payload in the body: 

`{
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 2,  # mode
            "param2": 0.0,  # custom mode
            "param3": 0.0,  # custom submode
            "param4": 0.0,  # empty
            "param5": 0.0,  # empty
            "param6": 0.0,  # empty
            "param7": 0.0,  # empty
            "command": {
                "type": "MAV_CMD_DO_SET_MODE"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}`

I can set only stabilize flight mode when changing param1: 1, 3, 5, 7, 9, 11 are changing the vehicle mode to stabilize which does not corresponds to the mode mapping i found in the code. Other values give me the previous error:

 Got MAVLink msg: COMMAND_ACK {command : 176, result : 3}

arendeutsch, 2020-08-26:
new update, it seems that to get it work param1 had to be 1 while param 2 is the mode mapping for the ardusub, i.e 1 - acro, 2 - depth, 19 - manual etc.

arendeutsch, 2020-08-26:
But maybe you can help me with some other issue. Im trying to send manual command with the following payload 
`    payload = {
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "MANUAL_CONTROL",
            "x": joy_x,
            "y": joy_y,
            "z": joy_z,
            "r": joy_r,
            "buttons": 0,
            "target": 0
        }
    }`

I do get OK 200 in my response but nothing happens. Can you try that command please ?

patrickelectric, 2020-08-26:
> I can set only stabilize flight mode when changing param1: 1, 3, 5, 7, 9, 11 are changing the vehicle mode to stabilize which does not corresponds to the mode mapping i found in the code. Other values give me the previous error:

Where you come up with this numbers ?
[MAV_CMD_DO_SET_MODE](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_MODE) uses [MAV_MODE](https://mavlink.io/en/messages/common.html#MAV_MODE) in param 1, as described in the documentation.
Like: 80 for MAV_MODE_STABILIZE_DISARMED, 208 for MAV_MODE_STABILIZE_ARMED and etc.
Besides that, I'm not aware if ardupilot supports MAV_MODE at all, the project uses mainly custom modes for each vehicle, so I recommend to set param1 as 1 and change param2 based in the custom mode that you want.
Right now there is a [problem in the ardusub](https://github.com/bluerobotics/ardusub-gitbook/issues/156) documentation that does not display the list of custom modes with the parameter value, but you can grab it from here: https://github.com/ArduPilot/ardupilot/blob/master/ArduSub/defines.h#L33
So you can use param1: 1 and param2: 0 for stabilize, or param2: 1 for acro and etc.
<details>
<summary>Examples</summary>
```
# Stabilize
'{                                                                                  
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 1,
            "param2": 0.0,
            "param3": 0.0,
            "param4": 0.0,
            "param5": 0.0,
            "param6": 0.0,
            "param7": 0.0,
            "command": {
                "type": "MAV_CMD_DO_SET_MODE"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}'

# Acro
'{                                                                                  
        "header": {
            "system_id": 1,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 1,
            "param2": 1,
            "param3": 0.0,
            "param4": 0.0,
            "param5": 0.0,
            "param6": 0.0,
            "param7": 0.0,
            "command": {
                "type": "MAV_CMD_DO_SET_MODE"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}'
```
</details>

patrickelectric, 2020-08-26:
> I do get OK 200 in my response but nothing happens. I was expecting the rov to run in 50% thrust forward Can you try that command please ?

You need to configure system_id to 255, since ardupilot will only accept this command from GCS (you can change this value in the GCS parameter ID), and target should be 1since this is the default sysid of ardupilot.

arendeutsch, 2020-08-27:
I've change the system id to 255 both in the header and in QGC (SYSID_THISMAV) but still nothong happens 

`{
        "header": {
            "system_id": 255,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "MANUAL_CONTROL",
            "x": 500,
            "y": 100,
            "z": 250,
            "r": 500,
            "buttons": 0,
            "target": 1
        }
    }`

Are there other parameters in QGS that need to be changed ? is this command should be sent in a some interval, I've tried different intervals of sending this request with no luck

patrickelectric, 2020-08-27:
Did you arm the vehicle before sending manual contorl ?

patrickelectric, 2020-08-28:
@arendeutsch Let me know if you have fixed your problem, from what I can tell just arming the vehicle and sending the manual control with the correct parameters should work

arendeutsch, 2020-08-30:
Sorry for the late replay @patrickelectric,

Everything works fine with sending manual command, i had to change a parameter in QGS.
Thank you very much for the support :+1:

patrickelectric, 2020-08-30:
thank you for the update

closed 2020-08-30 by patrickelectric