mavlink/mavlink2rest #23: tilting camera

arendeutsch, 2020-10-07:
Hi
I was wondering if its possible to tilt the camera with mavlink2rest.
I tried setting a tilt angle with the following body in the request 2500 centidegrees
                "header": {{
                    "system_id": 1,
                    "component_id": 1,
                    "sequence": 0
                }},
                "message": {{
                    "type": "COMMAND_LONG",
                    "param1": 1,
                    "param2": 2500,
                    "param3": 0,
                    "param4": 0,
                    "param5": 0,
                    "param6": 0,
                    "param7": 0,
                    "command": {{
                        "type": "MAV_CMD_DO_MOUNT_CONTROL"
                    }},
                "target_system": 0,
                "target_component": 0,
                "confirmation": 0

But i dont get any reaction.

patrickelectric, 2020-10-08:
Hi @arendeutsch, what kind of camera or vehicle are you using ? Does this camera support such command ?

arendeutsch, 2020-10-08:
Im testing on bluerov which its default camera, I would assume it supports it based on the fact that I can tilt the camera when im using QGroundControl

patrickelectric, 2020-10-08:
if you are using the bluerov, you'll need to change the last param (param7, mount mode) to [2 (MAV_MOUNT_MODE_MAVLINK_TARGETING)](https://mavlink.io/en/messages/common.html#MAV_MOUNT_MODE_MAVLINK_TARGETING), you can check the [message documentation here](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_MOUNT_CONTROL).

arendeutsch, 2020-10-08:
Thanks, I will try it and let you know how it goes

patrickelectric, 2020-10-13:
@arendeutsch any update ?

arendeutsch, 2020-10-14:
Sorry for the late replay @patrickelectric , I wasn't able to test it yet and wouldn't be able to do it before next week. But i will let you know as soon as i do it.

arendeutsch, 2020-10-19:
Hi @patrickelectric ,
I've tried setting param7 to 2, doesnt seem to help

patrickelectric, 2020-10-19:
try to change system_id to 255 to send the message as a ground station

patrickelectric, 2020-10-19:
Hi,

Can you try the following code ?
```
export DATA='{
        "header": {
            "system_id": 255,
            "component_id": 1,
            "sequence": 0
        },
        "message": {
            "type": "COMMAND_LONG",
            "param1": 4500.0,
            "param2": 0.0,
            "param3": 0.0,
            "param4": 0.0,
            "param5": 0.0,
            "param6": 0.0,
            "param7": 2.0,
            "command": {
                "type": "MAV_CMD_DO_MOUNT_CONTROL"
            },
            "target_system": 0,
            "target_component": 0,
            "confirmation": 0
      }
}'
curl --request POST http://0.0.0.0:8088/mavlink -H "Content-Type: application/json" --data $DATA
```

A warning: This message will not work under SITL but only in the real ROV or a real simulated ROV.

arendeutsch, 2020-10-20:
Im getting response OK 200 with empty body but nothing happens.
Does the rov have to armed ?

patrickelectric, 2020-10-20:
if you have a camera with a 3axis gimbal yes, the default bluerov only has pitch control.

closed 2020-10-25 by patrickelectric