mavlink/mavlink2rest #24: MANUAL_CONTROL command not supported

enderocean, 2020-10-18:
Hi,
I try to control BlueROV2 by sending MANUAL_CONTROL messages in POST mode to: 0.0.0.0:8088/mavlink
Message content is retrieved and adapted from [ helper ](http://0.0.0.0:8088/helper/message/MANUAL_CONTROL)
```
{
        "header": {
        "system_id": 1,
        "component_id":1,
        "sequence":0
        },
        "message": {
            "type":"MANUAL_CONTROL",
            "x":1000,
            "y":0,
            "z":0,
            "r":0,
            "buttons":0,
            "target":0
        }
}
```
mavlink2rest logout is two parenthesis as if everything was fine: ()
But command has no effect whatever the control mode I use.
Any ideas what happens ? Thanks
Note. Other commands in COMMAND_LONG are working well: ARM, DISARM, SET_MODE

patrickelectric, 2020-10-18:
Are you sending this message periodically ?
From ArduPilot documentation:
"When the autopilot is being commanded to move via RC_CHANNELS_RAW or MANUAL_CONTROL messages, the messages must be sent at a constant rate like the HEARTBEAT message. Otherwise, the autopilot will execute a failsafe if it has not received an updated command after a timeout period."

enderocean, 2020-10-18:
I send a HEARTBEAT message every second.
```
{
    "header":{
        "system_id":255,
        "component_id":0,
        "sequence":0
    },
    "message":{
        "type":"HEARTBEAT",
        "custom_mode":0,
        "mavtype":{
            "type":"MAV_TYPE_GENERIC"
        },
        "autopilot":{
            "type":"MAV_AUTOPILOT_GENERIC"
        },
        "base_mode":{"bits":128},
        "system_status":{
            "type":"MAV_STATE_UNINIT"
        },
        "mavlink_version":0
    }
}
```

enderocean, 2020-10-18:
I send a HEARTBEAT message every second. 
I will try to make MANUAL_CONTROL sent at a regular rate and let you know.

enderocean, 2020-10-18:
I have set a program that send :

- a HEARTBEAT msg every second
- a MANUAL_CONTROL every 200ms 4 second after the beginning. 

Normally, we are not in a "failsafe" case. But it does not work neither. The issue might be elsewhere.
I will close this one as long as mavlink2rest does the job and is not the root cause. Thank you.

closed 2020-10-18 by enderocean

enderocean, 2020-10-19:
I found the bug. Variable "system_id" must be set to 255 and "target" to 1 in MANUAL_CONTROL message.