mavlink/mavlink2rest #33: Ardusub command MAV_CMD_DO_SET_MODE is unsupported

enderocean, 2021-02-21:
Hi,
Below message is submitted on "0.0.0.0:8088/mavlink" URL and works well using SITL autotest tool
```
msg.payload = {
      "header": {
        "system_id": 1,
        "component_id": 1,
        "sequence": 0
      },
      "message": {
        "type":"COMMAND_LONG",
        "param1":19.0,
        "param2":0.0,"param3":0.0,"param4":0.0,"param5":0.0,"param6":0.0,"param7":0.0,
        "command":{
          "type":"MAV_CMD_DO_SET_MODE"
        },
        "target_system":0,
        "target_component":0,
        "confirmation":0
      }
    };
```
In this case, system is set in STABILIZE mode.

But same message does not work on real BlueROV2. 

When opening "0.0.0.0:8088/mavlink", I found this:
```
...
"COMMAND_ACK":{"command":{"type":"MAV_CMD_DO_SET_MODE"},"message_information":{"counter":3,"frequency":0.11999999731779099,"time":{"first_message":"2021-02-21T19:28:35.290372244+01:00","last_message":"2021-02-21T19:29:00.714305563+01:00"}},"progress":0,"result":{"type":"MAV_RESULT_UNSUPPORTED"},"result_param2":0,"target_component":0,"target_system":0,"type":"COMMAND_ACK"}
...
```
and
```
...
"HEARTBEAT":{"autopilot":{"type":"MAV_AUTOPILOT_ARDUPILOTMEGA"},"base_mode":{"bits":209},"custom_mode":19,"mavlink_version":3,"mavtype":{"type":"MAV_TYPE_SUBMARINE"},"message_information":{"counter":106,"frequency":1.009523868560791,"time":{"first_message":"2021-02-21T19:27:16.370862274+01:00","last_message":"2021-02-21T19:29:01.574418010+01:00"}},"system_status":{"type":"MAV_STATE_CRITICAL"},"type":"HEARTBEAT"}
...
```

Any idea why message is supported in SITL and not on BR2 ?
Thank you for your return.

patrickelectric, 2021-02-21:
Hi @enderocean, can you please inform your ardusub version and the entire output of the root page of the rest api ?

enderocean, 2021-02-21:
ArduSub Version: 3.5.4

patrickelectric, 2021-02-21:
Hi, please updates to the latest stable version (4.0.2)

enderocean, 2021-02-21:
Problem solved. Thank @patrickelectric

closed 2021-02-21 by enderocean