mavlink/mavlink2rest #34: Altitude not retrieved

enderocean, 2021-02-24:
Hi,
Using mavlink2rest with ardupilot autotest, altitude information is available in AHRS3 section. Data is updated regularly and correct.
Using physical BR2, this info is not available. Value is sticked to zero.
Any idea why ?
Tested with :
- ArduSub 4.0.2
- Companion software 0.0.22
On two different ROVs. 
Thank you for your return.

patrickelectric, 2021-02-24:
Hi,

How are you getting AHRS3 ? This message is not sent by SITL by default.
Do you have EKF3 enabled in SITL ? You can try to enable that in your ROV.
But, there are others attitude messages such as ATTITUDE and AHRS2 that you can use.

enderocean, 2021-02-24:
Here is the flow provided by mavlink2rest in autotest mode :

```
{"AHRS":{"accel_weight":0.0,"error_rp":0.0025952341966331005,"error_yaw":0.001920149428769946,"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.431172990+01:00","last_message":"2021-02-24T17:01:23.783130452+01:00"}},"omegaIx":-0.0004362220352049917,"omegaIy":-0.00036959859426133335,"omegaIz":-0.0005507777677848935,"renorm_val":0.0,"type":"AHRS"},"AHRS2":{"altitude":0.0,"lat":0,"lng":0,"message_information":{"counter":221,"frequency":4.018181800842285,"time":{"first_message":"2021-02-24T17:00:28.133524653+01:00","last_message":"2021-02-24T17:01:23.783578093+01:00"}},"pitch":0.001758397906087339,"roll":0.0019166297279298306,"type":"AHRS2","yaw":-1.566870927810669},"AHRS3":{"altitude":-0.17999999225139618,"lat":338103130,"lng":-1183938670,"message_information":{"counter":223,"frequency":3.982142925262451,"time":{"first_message":"2021-02-24T17:00:27.547465110+01:00","last_message":"2021-02-24T17:01:23.783793479+01:00"}},"pitch":0.0023309793323278427,"roll":0.0024154323618859053,"type":"AHRS3","v1":0.0,"v2":0.0,"v3":0.0,"v4":0.0,"yaw":-1.5667086839675903},"ATTITUDE":{"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.420496730+01:00","last_message":"2021-02-24T17:01:23.776817101+01:00"}},"pitch":0.0023309793323278427,"pitchspeed":0.0005453334888443351,"roll":0.0024154323618859053,"rollspeed":0.00040633100434206426,"time_boot_ms":58348,"type":"ATTITUDE","yaw":-1.5667086839675903,"yawspeed":0.0003865770995616913},"BATTERY_STATUS":{"battery_function":{"type":"MAV_BATTERY_FUNCTION_UNKNOWN"},"battery_remaining":99,"charge_state":{"type":"MAV_BATTERY_CHARGE_STATE_UNDEFINED"},"current_battery":0,"current_consumed":22,"energy_consumed":7,"id":0,"mavtype":{"type":"MAV_BATTERY_TYPE_UNKNOWN"},"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.433095093+01:00","last_message":"2021-02-24T17:01:23.785413608+01:00"}},"temperature":32767,"time_remaining":0,"type":"BATTERY_STATUS","voltages":[12587,65535,65535,65535,65535,65535,65535,65535,65535,65535]},"DISTANCE_SENSOR":{"covariance":0,"current_distance":0,"horizontal_fov":0.0,"id":0,"mavtype":{"type":"MAV_DISTANCE_SENSOR_UNKNOWN"},"max_distance":700,"message_information":{"counter":2,"frequency":null,"time":{"first_message":"2021-02-24T17:00:26.432454017+01:00","last_message":"2021-02-24T17:00:26.726191645+01:00"}},"min_distance":20,"orientation":{"type":"MAV_SENSOR_ROTATION_PITCH_270"},"quaternion":[0.0,0.0,0.0,0.0],"signal_quality":0,"time_boot_ms":2098,"type":"DISTANCE_SENSOR","vertical_fov":0.0},"EKF_STATUS_REPORT":{"airspeed_variance":0.0,"compass_variance":0.0015877227997407317,"flags":{"bits":895},"message_information":{"counter":223,"frequency":3.982142925262451,"time":{"first_message":"2021-02-24T17:00:27.550033578+01:00","last_message":"2021-02-24T17:01:23.784580696+01:00"}},"pos_horiz_variance":0.0008397054043598473,"pos_vert_variance":0.34894177317619324,"terrain_alt_variance":0.0,"type":"EKF_STATUS_REPORT","velocity_variance":0.014957706443965435},"GLOBAL_POSITION_INT":{"alt":-180,"hdg":27024,"lat":338103130,"lon":-1183938670,"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.423339689+01:00","last_message":"2021-02-24T17:01:23.777505739+01:00"}},"relative_alt":-72,"time_boot_ms":58348,"type":"GLOBAL_POSITION_INT","vx":1,"vy":2,"vz":-13},"GPS_GLOBAL_ORIGIN":{"altitude":-20,"latitude":338103130,"longitude":-1183938670,"message_information":{"counter":1,"frequency":null,"time":{"first_message":"2021-02-24T17:00:38.225659826+01:00","last_message":"2021-02-24T17:00:38.225664342+01:00"}},"time_usec":12823202,"type":"GPS_GLOBAL_ORIGIN"},"GPS_RAW_INT":{"alt":-280,"alt_ellipsoid":0,"cog":0,"eph":121,"epv":200,"fix_type":{"type":"GPS_FIX_TYPE_RTK_FIXED"},"h_acc":200,"hdg_acc":0,"lat":338103130,"lon":-1183938670,"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.429803216+01:00","last_message":"2021-02-24T17:01:23.782497888+01:00"}},"satellites_visible":10,"time_usec":58177000,"type":"GPS_RAW_INT","v_acc":200,"vel":0,"vel_acc":40,"yaw":0},"HEARTBEAT":{"autopilot":{"type":"MAV_AUTOPILOT_ARDUPILOTMEGA"},"base_mode":{"bits":81},"custom_mode":19,"mavlink_version":3,"mavtype":{"type":"MAV_TYPE_SUBMARINE"},"message_information":{"counter":59,"frequency":1.0350877046585083,"time":{"first_message":"2021-02-24T17:00:25.742405483+01:00","last_message":"2021-02-24T17:01:23.441211192+01:00"}},"system_status":{"type":"MAV_STATE_STANDBY"},"type":"HEARTBEAT"},"HOME_POSITION":{"altitude":-110,"approach_x":0.0,"approach_y":0.0,"approach_z":0.0,"latitude":338103130,"longitude":-1183938670,"message_information":{"counter":1,"frequency":null,"time":{"first_message":"2021-02-24T17:00:38.226441741+01:00","last_message":"2021-02-24T17:00:38.226446241+01:00"}},"q":[1.0,0.0,0.0,0.0],"time_usec":12823202,"type":"HOME_POSITION","x":0.0,"y":0.0,"z":0.0},"HWSTATUS":{"I2Cerr":0,"Vcc":5000,"message_information":{"counter":227,"frequency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.431797017+01:00","last_message":"2021-02-24T17:01:23.784002004+01:00"}},"type":"HWSTATUS"},"LOCAL_POSITION_NED":{"message_information":{"counter":6,"frequency":6.0,"time":{"first_message":"2021-02-24T17:01:22.513656853+01:00","l [... cut ...] requency":3.9824562072753906,"time":{"first_message":"2021-02-24T17:00:26.432917014+01:00","last_message":"2021-02-24T17:01:23.785158555+01:00"}},"time_usec":58348318,"type":"VIBRATION","vibration_x":0.005627060774713755,"vibration_y":0.005976642947643995,"vibration_z":0.00621375534683466}}
```
I understand AHRS3 is actually activated on autotest.
When I analyze the mavlink2rest flow on BR2, altitude values in ATTITUDE and AHRS2 are not set. So I can not use it.
How do you activate EKF3 on BR2 ?

patrickelectric, 2021-02-24:
> When I analyze the mavlink2rest flow on BR2, altitude values in ATTITUDE and AHRS2 are not set. So I can not use it.

How so ? They should be populated, if not that's a huge problem. please provide the output of mavlink2rest of your ROV, you should also test it under SITL.

I just tested it here and it works:
[image]

> How do you activate EKF3 on BR2 ?
Check the EKF parameters here: https://www.ardusub.com/developers/full-parameter-list.html

enderocean, 2021-02-24:
I will do some test and let you know. Thank you.

newmips, 2021-03-03:
Sorry for late answer. I was in travel...
I have set AHRS_EFK_TYPE to type 3 with QGC. But result is the same. Altitude is never valued.
Here is the mavlink flow:
```
{"AHRS":{"accel_weight":0.0,"error_rp":0.8961499929428101,"error_yaw":0.84513920545578,"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.399978881+01:00","last_message":"2021-03-03T12:09:08.700585264+01:00"}},"omegaIx":-0.013802853412926197,"omegaIy":-0.00929949525743723,"omegaIz":0.011294032447040081,"renorm_val":0.0,"type":"AHRS"},"AHRS2":{"altitude":0.0,"lat":0,"lng":0,"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.401701799+01:00","last_message":"2021-03-03T12:09:08.702310443+01:00"}},"pitch":0.029463812708854675,"roll":0.0067618899047374725,"type":"AHRS2","yaw":-0.3136591911315918},"AHRS3":{"altitude":0.0,"lat":0,"lng":0,"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.403341453+01:00","last_message":"2021-03-03T12:09:08.706595736+01:00"}},"pitch":0.029463812708854675,"roll":0.0067618899047374725,"type":"AHRS3","v1":0.0,"v2":0.0,"v3":0.0,"v4":0.0,"yaw":-0.3136591911315918},"ATTITUDE":{"message_information":{"counter":1005,"frequency":10.050000190734863,"time":{"first_message":"2021-03-03T12:07:28.382076707+01:00","last_message":"2021-03-03T12:09:08.775305353+01:00"}},"pitch":0.8314077258110046,"pitchspeed":-0.0141063891351223,"roll":1.0058766603469849,"rollspeed":0.6315346360206604,"time_boot_ms":247894,"type":"ATTITUDE","yaw":2.6747241020202637,"yawspeed":-0.27128782868385315},"BATTERY_STATUS":{"battery_function":{"type":"MAV_BATTERY_FUNCTION_UNKNOWN"},"battery_remaining":97,"charge_state":{"type":"MAV_BATTERY_CHARGE_STATE_UNDEFINED"},"current_battery":593,"current_consumed":404,"energy_consumed":215,"id":0,"mavtype":{"type":"MAV_BATTERY_TYPE_UNKNOWN"},"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.411146635+01:00","last_message":"2021-03-03T12:09:08.724151866+01:00"}},"temperature":32767,"time_remaining":0,"type":"BATTERY_STATUS","voltages":[14796,65535,65535,65535,65535,65535,65535,65535,65535,65535]},"GLOBAL_POSITION_INT":{"alt":-10,"hdg":15325,"lat":0,"lon":0,"message_information":{"counter":1005,"frequency":10.050000190734863,"time":{"first_message":"2021-03-03T12:07:28.382461711+01:00","last_message":"2021-03-03T12:09:08.775343895+01:00"}},"relative_alt":-11,"time_boot_ms":247894,"type":"GLOBAL_POSITION_INT","vx":0,"vy":0,"vz":0},"GPS_RAW_INT":{"alt":0,"alt_ellipsoid":0,"cog":0,"eph":65535,"epv":65535,"fix_type":{"type":"GPS_FIX_TYPE_NO_GPS"},"h_acc":0,"hdg_acc":0,"lat":0,"lon":0,"message_information":{"counter":96,"frequency":0.9795918464660645,"time":{"first_message":"2021-03-03T12:07:29.102180677+01:00","last_message":"2021-03-03T12:09:08.011645028+01:00"}},"satellites_visible":0,"time_usec":0,"type":"GPS_RAW_INT","v_acc":0,"vel":0,"vel_acc":0,"yaw":0},"HEARTBEAT":{"autopilot":{"type":"MAV_AUTOPILOT_ARDUPILOTMEGA"},"base_mode":{"bits":81},"custom_mode":19,"mavlink_version":3,"mavtype":{"type":"MAV_TYPE_SUBMARINE"},"message_information":{"counter":101,"frequency":1.0099999904632568,"time":{"first_message":"2021-03-03T12:07:28.437286641+01:00","last_message":"2021-03-03T12:09:08.448437898+01:00"}},"system_status":{"type":"MAV_STATE_CRITICAL"},"type":"HEARTBEAT"},"HWSTATUS":{"I2Cerr":0,"Vcc":5103,"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.404986939+01:00","last_message":"2021-03-03T12:09:08.723151327+01:00"}},"type":"HWSTATUS"},"MEMINFO":{"brkval":0,"freemem":65535,"freemem32":111008,"message_information":{"counter":1005,"frequency":10.050000190734863,"time":{"first_message":"2021-03-03T12:07:28.384258996+01:00","last_message":"2021-03-03T12:09:08.776942668+01:00"}},"type":"MEMINFO"},"MISSION_CURRENT":{"message_information":{"counter":1005,"frequency":10.050000190734863,"time":{"first_message":"2021-03-03T12:07:28.385090810+01:00","last_message":"2021-03-03T12:09:08.780918492+01:00"}},"seq":0,"type":"MISSION_CURRENT"},"MOUNT_STATUS":{"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.407246704+01:00","last_message":"2021-03-03T12:09:08.724084279+01:00"}},"pointing_a":0,"pointing_b":0,"pointing_c":0,"target_component":0,"target_system":0,"type":"MOUNT_STATUS"},"NAMED_VALUE_FLOAT":{"message_information":{"counter":7029,"frequency":70.29000091552734,"time":{"first_message":"2021-03-03T12:07:28.325707041+01:00","last_message":"2021-03-03T12:09:08.725341159+01:00"}},"name":["I","n","p","u","t","H","o","l","d","\u0000"],"time_boot_ms":247810,"type":"NAMED_VALUE_FLOAT","value":0.0},"NAV_CONTROLLER_OUTPUT":{"alt_error":0.2897911071777344,"aspd_error":0.0,"message_information":{"counter":1005,"frequency":10.050000190734863,"time":{"first_message":"2021-03-03T12:07:28.384705204+01:00","last_message":"2021-03-03T12:09:08.779269030+01:00"}},"nav_bearing":153,"nav_pitch":47.6365966796875,"nav_roll":57.63274383544922,"target_bearing":0,"type":"NAV_CONTROLLER_OUTPUT","wp_dist":0,"xtrack_error":0.0},"PARAM_VALUE":{"message_information":{"counter":10,"frequency":0.1149425283074379,"time":{"first_message":"2021-03-03T12:07:29.841287688+01:00","last_message":"2021-03-03T12:08:57.031457344+01:00"}},"param_count":920,"param_id":["G","N","D","_","A","B", [... cut ...] g_2":0,"message_information":{"counter":1004,"frequency":10.039999961853027,"time":{"first_message":"2021-03-03T12:07:28.408943810+01:00","last_message":"2021-03-03T12:09:08.724112652+01:00"}},"time_usec":247807351,"type":"VIBRATION","vibration_x":0.15067043900489807,"vibration_y":0.18063734471797943,"vibration_z":0.12864017486572266}}
```
I will dig further... Anyway, it seems not to be an issue of mavlink2rest if it works by your side. 
I will close this issue except of you have another idea ?

patrickelectric, 2021-03-03:
> Sorry for late answer. I was in travel...
> I have set AHRS_EFK_TYPE to type 3 with QGC. But result is the same. Altitude is never valued.
> Here is the mavlink flow:
>
> ``` [...]

From your json, all attitude messages are there. What is the problem ?

newmips, 2021-03-03:
"altitude" attribute is always with value "0" ; even when the ROV is moving up an down.

patrickelectric, 2021-03-03:
Sorry, about that is necessary to use `VFR_HUD` or `GLOBAL_POSITION_INT`, Altitude information is not provided by AHRS

enderocean, 2021-03-03:
OK. Thx. I will rely on those values.

closed 2021-03-03 by enderocean

patrickelectric, 2021-03-03:
Just an update, I'll be working to improve mavlink2rest web functionalities soon related to the main webpage, any suggestion is much appreciated.