mavlink/mavlink2rest #35: Rework frontpage

patrickelectric, 2021-03-02:


enderocean, 2021-03-03:
What do you mean by frontpage ?

patrickelectric, 2021-03-03:
0.0.0.0:8088
[image]

enderocean, 2021-03-03:
Well. As it is a REST API landing page, I suggest to divide it in several tabs / parts:

- First tab is a page that is accessing real time to the services like you did. It's really helpful and enable to have a global view of possibilities

- Second tab is a standard documentation of the input/output using a dedicated tool (like swagger, docAPI or any other). Objective is to be able to enrich documentation easily and create client programmes. I also imagine a  quick engine checkup feature where you could detect abnormal situations ; but it is clearly time demanding :clock1030: 

- Last tab is a list of actions that are commonly executed in ArduSub, ArduCopter, Ardu... For instance for ArduSub: arm/disarm, go forward, backward, stabilize, hold position, open / close gripper... Those examples could be referenced in the second tabs to illustrate the typical usage of mavlink2rest services.