mavlink/mavlink2rest #41: Continus build => 404

Dronotique, 2021-04-27:
Hello,

There is no more access to continious builds, the link is broken https://github.com/patrickelectric/mavlink2rest/releases/download/master/mavlink2rest-arm-unknown-linux-musleabihf

thanks!

patrickelectric, 2021-04-27:
Thanks for reporting it, I'll take a look

patrickelectric, 2021-05-04:
Hi @Dronotique, I would recommend for now to use the latest 0.10.0 version:
https://github.com/patrickelectric/mavlink2rest/releases/tag/0.10.0

Is there something necessary in master branch for you ?
Also, be aware that I'm working to improve mavlink2rest to have a better integration with multi-components vehicles and the API may break, I can share some ideias about how will this end up if you are interested.

Dronotique, 2021-05-07:
Hi @patrickelectric ,

No problem, I'm using 0.10 version, but sometimes I'm testing continius built one, and I just want to inform the link was broke

patrickelectric, 2021-05-18:
Hi @Dronotique I have finished an almost final version of the next release:
https://github.com/patrickelectric/mavlink2rest/actions/runs/854633845

It heavily changes the API and it's not the final version, but be free to take a look.

Dronotique, 2021-05-18:
OK, I think I could have time next wwek

patrickelectric, 2026-11-02:
Fixed in t0.11.13, binaries are under the latest release. https://github.com/mavlink/mavlink2rest/releases/tag/t0.11.13

closed 2026-11-02 by patrickelectric