mavlink/mavlink2rest #48: Error messages appearing when using tcpout connection string

dwaiting, 2022-02-12:
Getting lots of error messages when using a tcpout connection:

```
$ ./mavlink2rest-x86_64-unknown-linux-musl -c tcpout:127.0.0.1:14550
[2022-02-12T21:58:09Z ERROR mavlink2rest::mavlink_vehicle] Recv error: Io(Os { code: 11, kind: WouldBlock, message: "Resource temporarily unavailable" })
[2022-02-12T21:58:10Z ERROR mavlink2rest::mavlink_vehicle] Recv error: Io(Os { code: 11, kind: WouldBlock, message: "Resource temporarily unavailable" })
[2022-02-12T21:58:10Z ERROR mavlink2rest::mavlink_vehicle] Recv error: Io(Os { code: 11, kind: WouldBlock, message: "Resource temporarily unavailable" })
[2022-02-12T21:58:10Z ERROR mavlink2rest::mavlink_vehicle] Recv error: Io(Os { code: 11, kind: WouldBlock, message: "Resource temporarily unavailable" })
```

I don't get these error messages when using UDP or tcpin.

The application still appears to be working despite the error messages.

patrickelectric, 2022-02-25:
I saw it now, will check it asap

dwaiting, 2022-03-25:
I believe I've found the source of this error message in https://github.com/mavlink/rust-mavlink/blob/master/src/connection/tcp.rs#L33

`socket.set_read_timeout(Some(Duration::from_millis(100)))?;`

If I comment out the line above, the error message goes away and mavlink2rest still seems to work ok. Do you foresee any unintended consequences from this approach?

khancyr, 2023-08-25:
I can confirm it happens too.
Using WSL2 ubuntu22.04 on a fresh build
I didn't had it yesterday with the same build I think, but I change the SITL build in the mean time so no idea what can cause this for now.

ccskevin32, 2024-05-17:
Experienced this as well and reported as issue #102. Noticed this issue for what is probably the same thing when spending a little more time looking at the project. Running raspbian on a pi 0

closed 2026-11-02 by patrickelectric