mavlink/mavlink2rest #49: Getting 404 response for MAV_CMD_FIXED_MAG_CAL_YAW

dwaiting, 2022-03-24:
I get a 404 "Failed to parse image" response when sending the request below:

```
{
    "header": {
        "system_id": 255,
        "component_id": 0,
        "sequence": 0
    },
    "message":{
        "type": "COMMAND_LONG",
        "param1": 100.0,
        "param2": 0.0,
        "param3": 0.0,
        "param4": 0.0,
        "param5": 0.0,
        "param6": 0.0,
        "param7": 0.0,
        "command":{
            "type": "MAV_CMD_FIXED_MAG_CAL_YAW"
        },
        "target_system": 1,
        "target_component": 1,
        "confirmation":0
    }
}
```

I believe MAV_CMD_FIXED_MAG_CAL_YAW is part of ardupilotmega.xml so it should be supported. Is that correct?

(The MAV_CMD_COMPONENT_ARM_DISARM command is working great so everything else is set up correctly.)

Any assistance is appreciated!

dwaiting, 2022-03-24:
I see that MAV_CMD_FIXED_MAG_CAL_YAW was moved to common.xml about 2 years ago. However, the rust-mavlink package uses a commit of the Mavlink XML files from before this date, so the message still appears in ardupilotmega. 

I'm not sure if I should be reporting this as a problem with rust-mavlink or mavlink2rest.

patrickelectric, 2022-03-25:
This is a problem with rust-mavlink, it's necessary to update the mavlink version that's using

closed 2024-01-11 by patrickelectric