mavlink/mavlink2rest #52: NAMED_VALUE_FLOAT is invalid message

1chick, 2022-07-14:
pi@raspberrypi:~/companion $ curl --request POST http://0.0.0.0:8088/mavlink -H "Content-Type: application/json" --data '{ 
  "header": {
    "system_id": 255,
    "component_id": 0,
    "sequence": 0
  },
  "message": {
    "type":"NAMED_VALUE_FLOAT",
    "time_boot_ms":0,
    "name":"leak",
    "value":0.0
  }
}'
Failed to parse message, not a valid MAVLinkMessage.

1chick, 2022-07-15:
# message changed to it ,getcha
'{
  "header": {
    "system_id": 255,
    "component_id": 0,
    "sequence": 0
  },
  "message": {
    "type": "NAMED_VALUE_FLOAT",
    "time_boot_ms": 0,
    "value": 2.0,
    "name": [
      "l",
      "e",
      "a",
      "k",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000"
    ]
  }
}'

closed 2022-07-15 by 1chick

patrickelectric, 2022-07-15:
Hi @1chick,

I would recommend to check the `helper` endpoint for message example:
```
 curl http://blueos.local:6040/helper/mavlink\?name\=NAMED_VALUE_FLOAT                               
{
  "header": {
    "system_id": 255,
    "component_id": 0,
    "sequence": 0
  },
  "message": {
    "type": "NAMED_VALUE_FLOAT",
    "time_boot_ms": 0,
    "value": 0.0,
    "name": [
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000",
      "\u0000"
    ]
  }
}%                                                       
```

1chick, 2022-07-27:
hi bro @patrickelectric 
when post data，it response slowly，sometime need 40s. how to Optimize？