mavlink/mavlink2rest #65: COMMAND_INT message failing to parse

ericjohnson97, 2023-03-22:
First, this is a fantastic library and thank for all your hard work creating and maintaining it. 

I am trying to command a px4 drone, and I am able to arm and takeoff using COMMAND_LONG, but for movement I would like to use COMMAND_INT. I have tried a bunch of different variations of this, but I cannot get the below to parse

```
{
    "header": {
        "system_id": 255,
        "component_id": 240,
        "sequence": 0
    },
    "message": {
        "type":"COMMAND_INT",
        "target_system": 1,
        "target_component": 1,
        "frame": 1,
        "current" : 0,
        "autocontinue": 0,
        "command": {
            "type": "MAV_CMD_DO_REPOSITION"
        },
        "param1": 0.0,
        "param2": 0.0,
        "param3": 0.0,
        "param4": 0.0,
        "x": 476132388,
        "y": 1223124133,
        "z": 100.0
    }
}
```
The error mavlink to rest returns is 
```
Failed to parse message, not a valid MAVLinkMessage.
```

patrickelectric, 2023-03-22:
you are creating the message in a wrong format, frame is an enum.
```js
{
    "header": {
        "system_id": 255,
        "component_id": 240,
        "sequence": 0
    },
    "message": {
        "type":"COMMAND_INT",
        "target_system": 1,
        "target_component": 1,
        "frame": { 
           "type": "MAV_FRAME_GLOBAL"
        },                
        "current" : 0,
        "autocontinue": 0,                 
        "command": {
            "type": "MAV_CMD_DO_REPOSITION"
        },            
        "param1": 0.0,
        "param2": 0.0,
        "param3": 0.0, 
        "param4": 0.0,  
        "x": 476132388,
        "y": 1223124133,
        "z": 100.0
    }
}
```
To check the message format, use the helper endpoint:
http://0.0.0.0:8088/helper/mavlink?name=COMMAND_INT

ericjohnson97, 2023-03-23:
Awesome! not sure why I didn't think to use the helper last night. 

Got it working now. Thanks!

closed 2023-03-23 by ericjohnson97