mavlink/mavlink2rest #68: how can i post the message to vehicle

ahammedshabas, 2023-04-13:
POST /mavlink. Sends the message to a specific vehicle.

patrickelectric, 2023-04-28:
I'm not sure if I follow, did you check the examples and the README ?
You should just send a POST or use websocket.

patrickelectric, 2023-05-13:
Please, let me know if you have further problems.

closed 2023-05-13 by patrickelectric

ahammedshabas, 2023-05-15:
i tried with post by message structure
  const armData = {
            "header": {
                "system_id": 1,
                "component_id": 1,
                "sequence": 0
            },
            "message": {
                "type": "COMMAND_LONG",
                "param1": 1.0,
                "param2": 0.0, "param3": 0.0, "param4": 0.0, "param5": 0.0, "param6": 0.0, "param7": 0.0,
                "command": {
                    "type": "MAV_CMD_COMPONENT_ARM_DISARM"
                },
                "target_system": 1,
                "target_component": 1,
                "confirmation": 1
              }
        };
        fetch('http://192.168.2.2:4777/mavlink', {
            method: 'POST',
            headers: {
                'Content-Type': 'application/json'
            },
            body: JSON.stringify(armData)
        })
            .then(() => {
                console.log('Arm sent to server', armData);
            })
            .catch((error) => {
                console.error('Error sending Arm to server:', error);
            });
    }
  }
  
  can i get any example of using websocket