mavlink/mavlink2rest #72: Minimal test

clydemcqueen, 2023-05-01:
Can I use mavlink2rest to view a very simple message stream, e.g., just HEARTBEAT messages?

I am sending HEARTBEAT msgs via pymavlink:

```
import time
from pymavlink import mavutil

conn1 = mavutil.mavlink_connection('udpout:localhost:14550', source_system=1, source_component=1)

while True:
    conn1.mav.heartbeat_send(12, 3, 128, 21, 3)
    time.sleep(0.5)
```

When I try to recieve these using mavlink2rest nothing shows up:
```
./mavlink2rest-x86_64-unknown-linux-musl --verbose --system-id 255 --component-id 0 --connect udpin:0.0.0.0:14550
```

Is there some handshake that I am missing?

patrickelectric, 2023-05-13:
Hi @clydemcqueen thanks for checking it! 
I was able to replicate the problem and will take a look into it. I was in a conference, thanks for your patience.

patrickelectric, 2023-05-13:
Hi @clydemcqueen, from my tests I got it working while using mavlink2rest with mavlink 1, the default for the project is 2.

```sh
./mavlink2rest --verbose --connect udpin:127.0.0.1:14550 --mavlink 1
```

clydemcqueen, 2023-05-21:
@patrickelectric Thanks so much for tracking this down!

I was able to set mavlink wire protocol 2 messages by setting an environment variable:

```
import os
os.environ['MAVLINK20'] = '1'
from pymavlink import mavutil
```

A little funky (triggers E402 error from pycodestyle.py) but AFAICT this is the only supported way to specify mavlink wire protocol 2.

I'll close this out.

closed 2023-05-21 by clydemcqueen

patrickelectric, 2023-05-21:
Hi Clyde, thanks for letting me know!