mavlink/mavlink2rest #73: Unable to build master: xml-rs dependency failed

dbaldwin, 2023-05-13:
Hi, first off this is a great and much needed project. I'm new to Rust but hope to get more involved with the project. That being said, I just pulled master and am running into a dependency error when building:

```
(base) db@db-ubuntu:~/_dev/mavlink2rest$ cargo build
warning: unused manifest key: package.path
    Updating crates.io index
error: failed to select a version for the requirement `xml-rs = "^0.2"`
candidate versions found which didn't match: 0.8.10, 0.8.9, 0.8.8, ...
location searched: crates.io index
required by package `mavlink v0.10.0`
    ... which satisfies dependency `mavlink = "^0.10.0"` of package `mavlink2rest v0.11.14 (/home/db/_dev/mavlink2rest)`
```

When looking at crates.io it appears xml-rs 0.2 is no longer available:

https://crates.io/crates/xml-rs/versions

I've tried messing around with different mavlink versions but then end up with compilation errors. I figured I would open a ticket and see if anyone can help shed some light on this.

Many thanks,
Dennis

patrickelectric, 2023-05-13:
which version of rust are you using ? the warning message appears to show that there is something wrong.

to install rust, please remove rust/cargo if installed via your distro package manager and use the official way: https://rustup.rs/

patrickelectric, 2023-05-13:
Hey @dbaldwin nice software solution with droneblocks! Check https://github.com/bluerobotics/BlueOS-docker, it would be really nice to have your project as a BlueOS extension!

patrickelectric, 2023-05-13:
> which version of rust are you using ? the warning message appears to show that there is something wrong.
> 
> to install rust, please remove rust/cargo if installed via your distro package manager and use the official way: https://rustup.rs/

In the end I was able to replicate your issue here locale, will update the project with a fix.

dbaldwin, 2023-05-13:
> Hey @dbaldwin nice software solution with droneblocks! Check https://github.com/bluerobotics/BlueOS-docker, it would be really nice to have your project as a BlueOS extension!

I will definitely check it out asap!

> In the end I was able to replicate your issue here locale, will update the project with a fix.

Thanks so much. If I can provide any other details please let me know. Keep up the great work.

patrickelectric, 2023-05-13:
@dbaldwin can you check the master branch now ? I'll tag and release a new version as well. A binary should be available on the hour.

dbaldwin, 2023-05-14:
@patrickelectric you nailed it :100: . Was able to build and run. At some point I'll have a better understanding of Rust and will be able to contribute. Thank you.

closed 2023-05-14 by dbaldwin