mavlink/mavlink2rest #76: 404 on all requests after default install

danielhonies, 2023-05-30:
tried to install on Ubuntu 22 with cargo. Running PX4 with gazebo sitl. After default start with verbose, cmd line reports that different messages are received. However sending requests through browser as well as through Postman always returns 404. 
Not sure how to debug further.

danielhonies, 2023-05-30:
update: http://0.0.0.0:8088/v1/mavlink/ATTITUDE returns 404 while http://0.0.0.0:8088/mavlink/vehicles/1/components/1/messages/ATTITUDE returns the expected message. Is the readme out of date?

patrickelectric, 2023-06-20:
Hi @danielhonies. We need to update the cargo releases, the latest update was 2 years ago. Please use the project binaries for now.