mavlink/mavlink2rest #80: MAVLink2 payloads have trailing zeros

clydemcqueen, 2023-07-10:
The MAVLink2 spec requires senders to implement payload truncation: https://mavlink.io/en/guide/serialization.html#payload_truncation

For example, calling the API:
~~~
curl --verbose http://127.0.0.1:6040/mavlink -H "accept: application/json" --data \
'{
    "header": {
        "system_id": 1,
        "component_id": 194,
        "sequence": 52
    },
    "message": {
        "type": "DISTANCE_SENSOR",
        "time_boot_ms": 201509,
        "min_distance": 20,
        "max_distance": 5000,
        "current_distance": 2325,
        "mavtype": {"type": "MAV_DISTANCE_SENSOR_ULTRASOUND"},
        "id": 1,
        "orientation": {"type": "MAV_SENSOR_ROTATION_PITCH_270"},
        "covariance": 255,
        "horizontal_fov": 0.52,
        "vertical_fov": 0.52,
        "quaternion": [0, 0, 0, 0],
        "signal_quality": 0
    }
}'
~~~

causes this MAVLink message to be sent via UPD:
~~~
0000   fd 27 00 00 3b 01 c2 84 00 00 25 13 03 00 14 00
0010   88 13 15 09 01 01 19 ff b8 1e 05 3f b8 1e 05 3f
0020   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0030   00 f8 1a
~~~

The last 2 bytes are the CRC, the preceding 17 bytes (quaternion and signal_quality fields) should have been truncated before the CRC was calculated.

This may be the root cause of this bug: https://github.com/bluerobotics/BlueOS-docker/issues/1740

patrickelectric, 2023-08-07:
Source: https://github.com/mavlink/rust-mavlink/issues/188

patrickelectric, 2026-11-02:
Fixed by https://github.com/mavlink/mavlink2rest/pull/134, payload truncation is done by rust-mavlink now. Not available in 1.0.0, will be in the next release.

closed 2026-11-02 by patrickelectric