mavlink/mavlink2rest #82: May I get some help with POSITION_TARGET_LOCAL_NED

dbaldwin, 2023-08-19:
Hi,

I'm a big fan of this project and trying to build a proof of concept GCS using Vue.js. I'm very familiar with MAVSDK and can control my drone via SITL using drone.offboard.set_position_ned. I'd like to experiment with sending local coordinates using mavlink2rest. I'm using curl to post the following position message:

```
{
  "header": {
    "system_id": 255,
    "component_id": 1,
    "sequence": 0
  },
  "message": {
    "type": "POSITION_TARGET_LOCAL_NED",
    "time_boot_ms": 0,
    "x": 5,
    "y": 5,
    "z": -5,
    "vx": 0.0,
    "vy": 0.0,
    "vz": 0.0,
    "afx": 0.0,
    "afy": 0.0,
    "afz": 0.0,
    "yaw": 0.0,
    "yaw_rate": 0.0,
    "type_mask": {
      "bits": 0
    },
    "coordinate_frame": {
      "type": "MAV_FRAME_LOCAL_NED"
    },
    "target_system": 1,
    "target_component": 1,
    "confirmation": 0
  }
}
```

I get an Ok. response but the drone does not move when viewing in QGC. I'm wondering if someone may be able to help review my JSON request and see if you notice anything that I'm missing or doing work?

Thank you.

patrickelectric, 2024-01-11:
Hey @dbaldwin are you still having such problems ?

dbaldwin, 2024-01-11:
@patrickelectric I got this working but can't remember if I took another approach. I will close it for now and when I get back to it I will update the issue with what I did. Thanks for checking in.

closed 2024-01-11 by dbaldwin