mavlink/mavlink2rest #92: Compilation Error on M2 MacOS 14

neurons9, 2024-01-09:
Compiling ioctl-rs v0.1.6
error[E0277]: cannot subtract `chrono::Duration` from `Tm`
   --> /Users/joa/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-http-1.0.1/src/cookie/jar.rs:224:44
    |
224 |             cookie.set_expires(time::now() - Duration::days(365));
    |                                            ^ no implementation for `Tm - chrono::Duration`
    |
    = help: the trait `Sub<chrono::Duration>` is not implemented for `Tm`
    = help: the following other types implement trait `Sub<Rhs>`:
              <Tm as Sub<time::Duration>>
              <Tm as Sub>

   Compiling lock_api v0.3.4
error[E0308]: mismatched types
   --> /Users/joa/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-http-1.0.1/src/cookie/mod.rs:736:40
    |
736 |         self.set_expires(time::now() + twenty_years);
    |                                        ^^^^^^^^^^^^ expected `time::Duration`, found `chrono::Duration`
    |
    = note: `chrono::Duration` and `time::Duration` have similar names, but are actually distinct types
note: `chrono::Duration` is defined in crate `chrono`
   --> /Users/joa/.cargo/registry/src/index.crates.io-6f17d22bba15001f/chrono-0.4.31/src/duration.rs:55:1
    |
55  | pub struct Duration {
    | ^^^^^^^^^^^^^^^^^^^
note: `time::Duration` is defined in crate `time`
   --> /Users/joa/.cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.1.45/src/duration.rs:45:1
    |
45  | pub struct Duration {
    | ^^^^^^^^^^^^^^^^^^^

how can I fix this? And what file type are the binaries for macOS in the releases?

patrickelectric, 2024-01-09:
Hi @neurons9, this is pretty weird, mac M* family should be aarch64, there is a binary available on our release CI if you want to try.

How did you install Rust in your computer ? Did you follow [**the book**](https://doc.rust-lang.org/book/) ?

neurons9, 2024-01-09:
Hi @patrickelectric, thanks for your quick reply!
I have installed Rust with the command:
curl https://sh.rustup.rs -sSf | sh
What file types are the binaries? I can't open or run them...

patrickelectric, 2024-01-11:
I only have a mac m1 but I can compile fine here, from the issues that you are pointing, may be a dependence break or something.
Try to cargo clean, cargo update and cargo build

landswellsong, 2024-09-30:
@patrickelectric got a similar issue on Linux. Tried to repro on:
1. Gentoo (rustc version 1.71.1)
2. Docker `rust` (rustc version 1.81.0) clean repo
3. Ubuntu (rustc version 1.75.0)

I'm not a Rust coder so I have a limited way of debugging the build, but the underlying issue seems to be some change done in `chrono` library, as referenced in actix/actix-web#3135

Specifically this: https://github.com/chronotope/chrono/releases/tag/v0.4.30

I tried pinning it to `0.4.29`, but for some reason `cargo install` still pulls the latest one whereas `cargo tree` does understand what I want. Apparently upgrading actix to v4 might also help, but that sounds terrifying for me.

patrickelectric, 2024-09-30:
Hi @landswellsong I was able to build using ubuntu 22.04 from a fresh installation:

root@7ab75cc00ae8:/mavlink2rest# history
    1  apt update -y
    2  apt install -y curl
    3  curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
    4  . "$HOME/.cargo/env"
    5  apt install -y git g++ binutils
    7  git clone https://github.com/mavlink/mavlink2rest
    8  cd mavlink2rest/
    9  cargo build
   10  history

landswellsong, 2024-10-01:
Hello @patrickelectric 

This does work. I think `cargo install` , `cargo build` and `cargo tree` all use a different dependency logic? I was using `install` as specified in the README.

That being said it seems that `actix-web` is pulled in twice:
```
   Compiling actix-web v4.9.0
   Compiling hyper v0.14.30
   Compiling actix-files v0.6.6
   Compiling actix-http v2.2.2
   Compiling hyper-rustls v0.24.2
   Compiling reqwest v0.11.27
   Compiling git2 v0.16.1
   Compiling awc v2.0.3
   Compiling actix-web v3.3.3
```

May there be a merit of bumping it to v4 in this project's dependencies? Seems like that would solve this issue and apparently `mavlink2rest` already can work with v4.

patrickelectric, 2024-10-01:
Yes, that would be possible, but timewise is tricky. At the moment I'm working on other projects, but be free to collaborate with it.

patrickelectric, 2026-11-02:
The error comes from actix-web 2, used by the 0.10.0 version in crates.io. Build from master or use the binaries from 1.0.0. https://github.com/mavlink/mavlink2rest/releases/tag/1.0.0

closed 2026-11-02 by patrickelectric