mavlink/mavlink2rest #100: Multiple Filters on Websocket

ericjohnson97, 2024-03-21:
Is mavlink to rest setup to accept multiple filters or is it just one message per websocket stream? 

I have tried this
```
"ws://192.168.1.222:6040/ws/mavlink?filter=LOCAL_POSITION_NED,ATTITUDE
```
but that doesn't seem to work

patrickelectric, 2024-03-21:
> Is mavlink to rest setup to accept multiple filters or is it just one message per websocket stream?
> 
> I have tried this
> 
> ```
> "ws://192.168.1.222:6040/ws/mavlink?filter=LOCAL_POSITION_NED,ATTITUDE
> ```
> 
> but that doesn't seem to work

Hi @ericjohnson97, you should be using a regex: "LOCAL_POSITION_NED|ATTITUDE"
As explained here: https://github.com/mavlink/mavlink2rest?tab=readme-ov-file#websocket

ericjohnson97, 2024-04-03:
Forgot to respond to this. 

Yep it is regex! I need to read more carefully 🤦‍♂️

closed 2024-04-03 by ericjohnson97