mavlink/mavlink2rest #101: Unable to ARM the drone

sundaram123krishnan, 2024-04-26:
I am able to send POST request,
[image]
i can see the output as well, 
But the drone is not getting armed

patrickelectric, 2024-04-26:
Did you check this example ?

https://github.com/mavlink/mavlink2rest?tab=readme-ov-file#request-vehicle-to-be-armed

sundaram123krishnan, 2024-04-26:
yaa i did referred to that example only,
this is the request i sent
curl --request POST http://172.22.254.115:8088/v1/mavlink -H "Content-Type: application/json" --data \
'{
  "header": {
    "system_id": 1,
    "component_id": 1,
    "sequence": 0
  },
  "message": {
    "type":"COMMAND_LONG",
    "param1": 1.0,
    "param2": 0.0,"param3":0.0,"param4":0.0,"param5":0.0,"param6":0.0,"param7":0.0,
    "command": {
      "type": "MAV_CMD_COMPONENT_ARM_DISARM"
    },
    "target_system": 1,
    "target_component": 1,
    "confirmation": 1
  }
}' --verbose

sundaram123krishnan, 2024-04-26:
Thx for ur kind response, 
it works now

closed 2024-04-26 by sundaram123krishnan