mavlink/mavlink2rest #105: I got in trouble when I try to use MAV_CMD_DO_REPOSITION,  always return MAV_RESULT_UNSUPPORTED

dayleMG, 2024-05-20:
Hello, First Thank you for making  awesome API ! It is  really cool.
But I have only one problem
When I tried to  send this command 
```
{
  "header": { "system_id": 255, "component_id": 1, "sequence": 0 },
  "message": {
    "type": "COMMAND_INT",
    "target_system": 2,
    "target_component": 1,
    "frame": { "type": "MAV_FRAME_GLOBAL" },
    "current": 0,
    "autocontinue": 0,
    "command": { "type": "MAV_CMD_DO_REPOSITION" },
    "param1": 0.0,
    "param2": 0.0,
    "param3": 0.0,
    "param4": 0.0,
    "x": 388140540,
    "y": 1288516750,
    "z": 100.0
  }
}
```

This one  is  Command_ACK message 
[image]

Only this command doesn't work

Even I change vehicle mode for this command. 
```
{
  "header": {
    "system_id": 255,
    "component_id": 0,
    "sequence": 0
  },
  "message": {
    "type": "COMMAND_LONG",
    "param1": 1.0,
    "param2": 4.0,
    "param3": 3.0,
    "param4": 0.0,
    "param5": 0.0,
    "param6": 0.0,
    "param7": 0.0,
    "command": {
      "type": "MAV_CMD_DO_SET_MODE"
    },
    "target_system": 2,
    "target_component": 1,
    "confirmation": 1

  }
}
```

[image]

And then I  tried to use MAV_CMD_DO_REPOSITION. It doesn't work .
What do I mistake?? 
Thank you for  reading my question.

clydemcqueen, 2024-05-30:
What kind of vehicle are you sending this to? E.g., ArduSub does not support MAV_CMD_DO_REPOSITION.

dayleMG, 2024-05-31:
@clydemcqueen Hello, Thank you for relpying

I have used PX4-Gazebo simulator(headless) release of PX4, which is currently v1.14.2.

closed 2024-07-10 by dayleMG