mavlink/mavlink2rest #108: Connect to SITL

ShafSpecs, 2024-07-06:
Hello, I have got an SITL running on `udp:127.0.0.1:14550`, anytime I start mavlink2rest via: `.\mavlink2rest-x86_64-pc-windows-msvc.exe --verbose --connect udpin:127.0.0.1:14550`, I don't get any output. Is there a handshake protocol I am missing? Using this script though:
```py
import time
from pymavlink import mavutil

conn1 = mavutil.mavlink_connection('udpout:localhost:14551', source_system=1, source_component=1)

while True:
    conn1.mav.heartbeat_send(12, 3, 128, 21, 3)
    time.sleep(0.75)
```
I get the heartbeat messages.

And looking at the output (`/v1/mavlink`), vehicles is an array, so is there a way to connect (create) extra vehicles without running a separate process for that vehicle?

patrickelectric, 2024-07-06:
in this case you should be using `udpout` and not `udpin`, as in your python client. Did you tried that ?

ShafSpecs, 2024-07-06:
I managed to fix the issue of connection, but how do I connect extra vehicles? And looking at this issue: https://github.com/mavlink/mavlink2rest/issues/29, can the websocket receive commands?

ShafSpecs, 2024-07-06:
> in this case you should be using `udpout` and not `udpin`, as in your python client. Did you tried that ?

Yeah, I did. Fixed it :+1:

patrickelectric, 2024-07-06:
Hi @ShafSpecs you can send and receive from websocket, but at the moment you can communication across services that are also connected via websocket as well. To connect with multiple vehicles, I would recommend using mavlink-router as a middle-man, where it has all your network topology defined with an ednpoint to mavlink2rest

patrickelectric, 2024-07-06:
btw, if you are working in a groundcontrol station software, take a look in www.github.com/bluerobotics/cockpit

ShafSpecs, 2024-07-06:
Thank you. Any other main issues encountered would be handled in another issue.

> where it has all your network topology defined with an ednpoint to mavlink2rest

Regarding this though, it means I would simulate one connection via mavlink-router with mavlink2rest but in reality is receiving multiple inputs from several vehicles?

closed 2024-07-06 by ShafSpecs

ShafSpecs, 2024-07-06:
Looking at mavp2p, it doesn't allow connections on the fly. For example, I have one drone and I am connected to it via the router, then I want to add 2 more an hour later, and then another drone 4 hrs after that, how do I do that without disrupting the already running services?

patrickelectric, 2024-07-06:
Create a single UDP server with multiple clients, where each drone is a client.

ShafSpecs, 2024-07-06:
> Create a single UDP server with multiple clients, where each drone is a client.

Thanks, and if it's serial, I create a serial server? If I want to mix it though, like tcp + udp + serial? I think I would just create 3 servers for now. Thanks again, Patrick!

ShafSpecs, 2024-07-07:
> Create a single UDP server with multiple clients, where each drone is a client.

Would this also work for serial and tcp?