mavlink/mavlink2rest #109: Example command doesn't work

ShafSpecs, 2024-07-06:
Running the arned command present in the README: https://github.com/mavlink/mavlink2rest/blob/master/README.md#request-vehicle-to-be-armed, I get a parsing error in my terminal:
```sh
Failed to parse message, not a valid MAVLinkMessage.{
   "header": {
     "system_id": 255,
     "component_id": 240,
     "sequence": 0
   },
   "message": {
     "type":"COMMAND_LONG",
     "param1": 1.0,
     "param2": 0.0,"param3":0.0,"param4":0.0,"param5":0.0,"param6":0.0,"param7":0.0,
     "command": {
       "type": "MAV_CMD_COMPONENT_ARM_DISARM"
     },
     "target_system": 1,
     "target_component": 1,
     "confirmation": 1
   }}
```

patrickelectric, 2024-07-06:
Please provide the command that you are using to send the message, the output of it and the mavlink2rest log.

ShafSpecs, 2024-07-06:
> Please provide the command that you are using to send the message, the output of it and the mavlink2rest log.

This is the full command:
```sh
PS C:\Users\usr> curl --request POST http://127.0.0.1:8088/v1/mavlink -H "Content-Type: application/json" --data \
>> '{
>>    "header": {
>>      "system_id": 255,
>>      "component_id": 240,
>>      "sequence": 0
>>    },
>>    "message": {
>>      "type":"COMMAND_LONG",
>>      "param1": 1.0,
>>      "param2": 0.0,"param3":0.0,"param4":0.0,"param5":0.0,"param6":0.0,"param7":0.0,
>>      "command": {
>>        "type": "MAV_CMD_COMPONENT_ARM_DISARM"
>>      },
>>      "target_system": 1,
>>      "target_component": 1,
>>      "confirmation": 1
>>    }
>> }'
Failed to parse message, not a valid MAVLinkMessage.{
   "header": {
     "system_id": 255,
     "component_id": 240,
     "sequence": 0
   },
   "message": {
     "type":"COMMAND_LONG",
     "param1": 1.0,
     "param2": 0.0,"param3":0.0,"param4":0.0,"param5":0.0,"param6":0.0,"param7":0.0,
     "command": {
       "type": "MAV_CMD_COMPONENT_ARM_DISARM"
     },
     "target_system": 1,
     "target_component": 1,
     "confirmation": 1
   }
}
```

patrickelectric, 2024-07-06:
What is the output of mavlink2rest ?

ShafSpecs, 2024-07-07:
> What is the output of mavlink2rest ?

It doesn't output anything. If I pass the `--verbose` flag, only the telemetry messages gets outputed

ShafSpecs, 2024-07-07:
Re-ran the same command via postman and it worked, smh. Curl was failing. Probably a curl thing, then

closed 2024-07-07 by ShafSpecs