mavlink/mavlink2rest #114: Error message "Address Not Available"

VamsiTungala, 2024-09-20:
Hi 

I am new to Mavlink2rest api. I am trying to connect to device over LTE with private VPN. I was able to connect to device using QGCS but not to Mavlink2rest with the same IP address and port. I am getting following error.

`thread 'main' panicked at src/mavlink_vehicle.rs:43:71:
called `Result::unwrap()` on an `Err` value: Os { code: 99, kind: AddrNotAvailable, message: "Address not available" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
`
is it possible to connect to any other IP apart from local address?

patrickelectric, 2024-09-20:
Yes it is. Can you share the command line that you are using and how you configured your vehicle router ?

VamsiTungala, 2024-09-20:
Thank you for the response.
I figured out the issue. I was using udpin instead of udpout. now its working with udpout. 
`./mavlink2rest-x86_64-unknown-linux-musl --component-id 25 --connect udpout:<IP>:<PORT>`
Is this the correct way to do?

patrickelectric, 2024-09-20:
I'm glad that you figure out the problem, you should connect servers to clients indeed.

closed 2024-09-20 by patrickelectric